Nickduino/Pi-Somfy #94: Using Pi-Somfy with RFXCOM

engelrolling, 2020-09-14:
Hello,

I have a rfxtrx433e controller and want to use it to control my Somfy venetian blinds.

The pairing of the controller and the blind was done using their windows tool and now I'd like to use it through a raspberry. 

I did the tutorial and reached the point where I had to pair my transmitter with the blinds, but my blind didn't jog :/

Can I control my blinds through the transceiver I have using this repo?

Nickduino, 2020-09-14:
I've never used a RFXtrx433E, so I don't know if you can use it as a dumb emitter (because that's what we're doing here: the Python script runs the Somfy algorithm).

If I were you, I'd either sell my RFXtrx433E for 80€ and buy a cheap Chinese emitter + 433.42 MHz oscillators for 5€ **or** use the RFX controller the way it is intended with Home Automation, Domoticz or whatever. The fact you speak french is a bonus: many resources are in french

engelrolling, 2020-09-14:
Thanks for the fast answer. I found a library that might just do it. Didn't realize all this was french and indeed found much more ressources that way. Thanks for the tip :)

closed 2020-09-14 by engelrolling