Nickduino/Pi-Somfy #95: Web interface automatically starting again after a few hours

Cynwrig, 2020-09-29:
Dear Community,

I don't seem to find how to disable the Web interface on long term as I want to use the shell commands instead. 

My crontab file is empty and my systemctl is disabled inactive dead.

Thanks in advance

rbswift, 2020-09-30:
Presume you already did ... sudo systemctl disable shutters.service

What is the full command line you are using? -a starts the web interface so probably don't use that.

Cynwrig, 2020-09-30:
> Presume you already did ... sudo systemctl disable shutters.service
> 
> What is the full command line you are using? -a starts the web interface so probably don't use that.

Yep already did that, found the problem. The default crontab file under etc/systemctl is not the same one as when opened under crontab - e... So that one was still populated.

closed 2020-09-30 by Cynwrig