Nickduino/Pi-Somfy #96: erro launch operateshutters.conf

akuaman, 2020-10-20:
when lauch operateshutthers.conf i have this error

This program requires the modules located from the same github repository that a                                                                                        re not present.

Error: No module named 'requests'

KrejtSparck, 2021-01-15:
I triggered this exact same error om Python 2(!) (not on Python 3) and managed to solve the problem with the command "sudo pip install requests", after which a "sudo python /home/pi/Pi-Somfy/operateShutters.py" worked just fine. Strange that this went wrong on Python 2 at first, because nowhere the dependency on 'requests' is indicated (not in the requirements.txt or the README.md) and I don't see it used in the operateShutters.py code? I'm sure somebody (Nick?) can explain this....

Nickduino, 2021-01-15:
Well, Nick would advise to use Python 3 😬

closed 2021-10-10 by Nickduino