Nickduino/Pi-Somfy #101: Python 2 install using installation instructions throws Error: No module named requests

KrejtSparck, 2021-01-14:
I did a fresh install of Pi-Somfy on a freshly installed PI 3 B+ with Raspberry Pi OS Lite using the installation instructions to the letter. I opted for a Python 2 install. The very first use of operateShutters.py threw this error:
	This program requires the modules located from the same github repository that are not present.
	Error: No module named requests

What did I do:   (how to reproduce this)

sudo apt-get update
sudo apt-get install git
	0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.

In the /home/pi directory, type:

git clone https://github.com/Nickduino/Pi-Somfy.git

## Next, we need to install Python Libraries.

sudo apt-get install python-pip
	0 upgraded, 37 newly installed, 0 to remove and 0 not upgraded.

## Next, we need to install the PIGPIO libraries, to do so, type:

sudo apt-get install pigpio
	0 upgraded, 8 newly installed, 0 to remove and 0 not upgraded.

## Next install the required Python Libraries:

sudo pip install -r requirements.txt
	Successfully installed Flask-1.1.2 Jinja2-2.11.2 MarkupSafe-1.1.1 Werkzeug-1.0.1 click-7.1.2 ephem-3.7.7.1
		itsdangerous-1.1.0 paho-mqtt-1.5.1 pigpio-1.78

## Next, let's test if it all works. Start <operateShutters.py> by typing:

sudo python /home/pi/Pi-Somfy/operateShutters.py

	This program requires the modules located from the same github repository that are not present.
	Error: No module named requests

What did I do wrong, or what is wrong with the installation instructions?

rbswift, 2021-01-15:
Python 2 was sunset over a year ago. You should use python 3 as suggested in the install guide.

https://www.python.org/doc/sunset-python-2/

KrejtSparck, 2021-01-15:
With all respect, but allow me to quote form the README.md: "_Next, we need to install Python Libraries. Before doing so, you have to decide whether you want to run Pi-Somfy in Python 2 or Python 3. The library supports both, but Python 3 is suggested_". I was not aware the days are over for Python 2 (thanks for the info!) and gave that a try. If Python 2 is no longer supported by Pi-Somfy, it would be good to fix the README. (Or, if it should work fine on Python 2, my remarks might trigger somebody to look into the error I stumbled upon and fix something else, or the README.)

Nickduino, 2021-03-08:
You're welcome to fix the readme and do a pull request

closed 2021-03-08 by Nickduino