Nickduino/Pi-Somfy #103: CC1101

HerbOtto, 2021-01-22:
Has anyone successfully integrated the CC1101 programmable transmitter w/this library?

closed 2021-01-24 by HerbOtto

dckiller51, 2021-03-06:
hello, how did you implant the cc1101 cutter? I want to do the same for my father because he does not have a home automation system and this solution by pi seems ideal to me to be autonomous. Thank you

dckiller51, 2021-03-08:
> Has anyone successfully integrated the CC1101 programmable transmitter w/this library?
@HerbOtto 
hello, how did you implant the cc1101 cutter? I want to do the same for my father because he does not have a home automation system and this solution by pi seems ideal to me to be autonomous. Thank you

HerbOtto, 2021-03-08:
> @HerbOtto
>
> hello, how did you implant the cc1101 cutter? I want to do the same for my father because he does not have a home automation system and this solution by pi seems ideal to me to be [...]

I used the following library for the Pi...

https://github.com/fphammerle/python-cc1101

Worked quite well.

dckiller51, 2021-03-08:
> I used the following library for the Pi...
>
> https://github.com/fphammerle/python-cc1101
>
> Worked quite well.

Anglais

Thank you for your reply. Once the library is installed, I only change the TXGPIO: 4 to TXGPIO: 17. Or I have to add something else. As I am not at my father's house I would like to have all the information to do the tests.

HerbOtto, 2021-03-08:
The CC1101 is a programmable radio - you'll need to add code to setup the correct frequency [433.42 MHz] and assign TXGPIO to the GDO0 pin of the radio.  The radio is programmed via the CC1101's SI, SCLK, CSn pins using the above mention library.  

I used the following wiring scheme for the radio to raspberry pi...

```
# 	CC1101 -> Raspberry Pi (pin)
#
# 	VDD ---> 3.3V ---- (Pin 17)
#	GDO0 --> GPIO24 -- (Pin 18) [output]
# 	SI ----> MOSI ---- (Pin 19)
# 	GND ---> Ground -- (Pin 20)
# 	SCLK --> SCLK ---- (Pin 23)
# 	CSn ---> CE0 ----- (Pin 24)
```
I then edited operateShutters.py sendCommand()...

replaced...[line 316...]
```
    pi.wave_send_once(wid)
    while pi.wave_tx_busy():
        pass
```
with...
```
with cc1101.CC1101() as transceiver:
    transceiver.set_base_frequency_hertz(433.42e6)
    transceiver._write_burst(start_register=0x3E, values=[0x0, 0x34])    
    with transceiver.asynchronous_transmission():
        pi.wave_send_once(wid)
        while pi.wave_tx_busy():
            pass
```

Added some extra stuff...

```
import cc1101 # add to top of file

# Modify the Somfy() class: def __init__ ...
self.TXGPIO = 24 # assign to GPIO-24 as shown in above wiring
```

Hope this helps.

dckiller51, 2021-03-11:
@HerbOtto thank you for sharing it worked with my first try.

89jd, 2021-03-31:
I have followed steps above, but getting an error: -

raceback (most recent call last):
  File "operateShutters.py", line 563, in <module>
    MyShutter = operateShutters(args = args)
  File "operateShutters.py", line 398, in __init__
    self.ProcessCommand(args);
  File "operateShutters.py", line 459, in ProcessCommand
    self.shutter.program(self.config.ShuttersByName[args.shutterName])
  File "operateShutters.py", line 209, in program
    self.sendCommand(shutterId, self.buttonProg, 1)
  File "operateShutters.py", line 321, in sendCommand
    with transceiver.asynchronous_transmission():
  File "/usr/local/lib/python3.7/contextlib.py", line 112, in __enter__
    return next(self.gen)
  File "/home/pi/Pi-Somfy/.venv/lib/python3.7/site-packages/cc1101/__init__.py", line 937, in asynchronous_transmission
    self._set_transceive_mode(_TransceiveMode.ASYNCHRONOUS_SERIAL)
  File "/home/pi/Pi-Somfy/.venv/lib/python3.7/site-packages/cc1101/__init__.py", line 793, in _set_transceive_mode
    pktctrl0 = self._read_single_byte(ConfigurationRegisterAddress.PKTCTRL0)
  File "/home/pi/Pi-Somfy/.venv/lib/python3.7/site-packages/cc1101/__init__.py", line 219, in _read_single_byte
    response = self._spi.xfer([register | self._READ_SINGLE_BYTE, 0])
OSError: [Errno 9] Bad file descriptor

Looks to be part of cc1101 library, any ideas?

no1knows, 2022-05-26:
In case anyone else comes across this - I finally got it working after much ado on a RPi Zero by using the wiring shown here: https://github.com/SpaceTeddy/CC1101, i.e.:

```
CC1101<->Raspi
Vdd    -    3.3V (P1-01)
SI     -    MOSI (P1-19)
SO     -    MISO (P1-21)
CS     -    SS   (P1-24)
SCLK   -    SCK  (P1-23)
GDO2   -    GPIO (P1-22)
GDO0   -    not used in this demo [editor's note: I connected this to pin 18]
GND    -    P1-25
```
Also note that you need to enable SPI on the RPi by running `sudo raspi-config` and enabling SPI in the interfaces menu.

Also, worth defining `TXGPIO = 24` in operateShutters.conf since the code looks there first to define that output pin.

Big thanks to @HerbOtto for the code tweaks above (which worked nicely once the wiring was sorted).

Nickduino, 2022-05-27:
Great to hear it's working