Nickduino/Pi-Somfy #106: How to understand if I have soldered the crystal properly?

ybelakov, 2021-02-13:
After soldering I took my multimeter and tested every path on the emitter (continuity mode) and it seems that the paths are the same as in the original emitter (before desoldering). However it is not working. I checked my wiring 100 times and I am sure it is correct. I checked the jumper cables with the multimeter, too (they are OK). Any suggestions?

yavinmaster, 2021-02-14:
Do you have any normal RF switches you could test?  I tested my transmitter with https://github.com/milaq/rpi-rf and a RF switch before soldering on the new crystal.  I have just gone through the full setup and it all worked for me.

Nickduino, 2021-02-14:
You could probably do the test with the new oscillator, it's unlikely your receive is as picky as Somfy's frequency-wise

ybelakov, 2021-02-14:
So you are basically suggesting something like this:
1) Connect both the RF receiver and RF emitter to the RPI 
2) Try sending signals from the emitter to the receiver and see if it works.

I still have one untouched emitter (433.92) which I can test with.
Did I understand you, guys, correctly?

Nickduino, 2021-02-14:
You would need another RECEIVER, not emitter. I don't know if you can run them both on the Pi at the same time but you could try. On the top of my head, I think there is a Python port of VirtualWire.

The other solution is to use two Pis or a Pi and an arduino (which I did when first learning the radio stuff)

ybelakov, 2021-02-14:
Hi, 
Thanks for the respond!

As far as I have understood this on the left is the emitter and this on the right is the receiver:
[image]

I will try connecting them both to the same RPI (I have only one) and see if it works. Here's a wiring diagram I will be using if somebody's interested in testing, too.
[image]

Nickduino, 2021-02-14:
The long one is your receiver, correct

alonjr, 2021-08-01:
@ybelakov I assume that you found the problem. Can you share with us what it was and how you resolved it. I am also struggling to get the solution work with my somfy powered awning.
Tnx

Nickduino, 2021-08-02:
>  my somfy powered awning

Are you sure they're using the RTS protocol?

Nickduino, 2021-08-02:
No, it can be io on the more recent hardware

alonjr, 2021-08-02:
@Nickduino I believe that you are correct!
The user manual of the remote control shows  the model name as _**Situo 1 io**_. 
In that case, perhaps this is the reason I am not able to get Pi-Somfy to work.
Is there a solution that you are aware of that might work for me?
tnx

Nickduino, 2021-08-02:
io is a more modern and safer two-way protocol. You'll need one of the two Somfy boxes (or possibly a third party that does io?) to control your stuff

alonjr, 2021-08-02:
Thanks. While I will not have use of your project, at least I learned something new and had fun doing so.

Nickduino, 2021-08-02:
Great! Good luck with the rest

sfla, 2021-08-14:
I too am struggling with this (and I'm sure they use RTS). I don't have any fancy meters to check the power, and I'm completely new to both raspberry pi and soldering. Is there any form of software I can use to monitor or query the connection to the emitter to see if it's at least recognized as a connection?

Nickduino, 2021-08-14:
> Is there any form of software I can use to monitor or query the connection to the emitter to see if it's at least recognized as a connection?

It's not a peripheral that needs a driver or something, it doesn't need to be "recognized": the RPi doesn't know there an emitter (or nothing) connected there.

Now if you want to make sure your soldering is good and your wiring correct, your best chance is to make the system work with your blinds.

hourliert, 2021-10-01:
Hi,
I just went through the setup this afternoon. My first transmitter was screwed and I only realized it after desoldering + soldering everything. Had to desolder + resolder another one, glad I bought 5.
I have used a multimeter to debug it and essentially making sure that: the ground pin connect with the ground of the crystal, the 5V pin connects with the top pin of the crystal, and any other links are still working fine.

It's now working like a charm, super happy! Thanks again!

Nickduino, 2021-10-03:
> It's now working like a charm, super happy! Thanks again!

That's some great news right there!

closed 2021-10-10 by Nickduino