Nickduino/Pi-Somfy #107: MQTT interface appears to hang

yavinmaster, 2021-02-18:
I have only just got this system up and running and all appeared initially to go well.  The problem I am having is that the MQTT interface which I use to control the blinds from Home Assistant appears to hang after a period of time (duration yet to be confirmed).

The symptoms are that on initial startup, everything works - web control and home assistant control via MQTT.  The log file shows all of the MQTT commands correctly.  After a while home assistant control stops working.  If I check with an MQTT browser I can see that Home Assistant is still publishing the required topics but these are no longer visible in the operateshutters log file.  Control via the web interface still works however.

I have tried two methods of running the script - via cron and via a service.  Both have the same issue.  The cron method has the suggested restart every hour but this still doesn't appear to help.  If I manually restart the service or kill and restart the script then everything returns to being operational again.

Any suggestions on where to look next or what could be going wrong?

yavinmaster, 2021-02-19:
Think I'm starting to home in on the issue.  I appears to be related to my backup which is run every day in the small hours.  Looking at the mqtt logs, when mosquito restarts after the backup, everything else reconnects but pi-somfy doesn't for some reason.  Going to try a service stop before the backup and a service start after the backup and see if that fixes it.

rbswift, 2021-02-19:
Don’t restart mosquitto daily. It’s not necessary

yavinmaster, 2021-02-21:
I shut down all of the databases on the system prior to doing the backup.  I therefore have to restart mosquitto daily.
I have decided to drop using the MQTT interface with Pi-Somfy as I was getting intermittent results.  Currently I am just running the web interface and sending curl commands to operate the shutters from home assistant.  It means I don't track the blind position, but as I can't stop the manual remote being used this wasn't of much use anyway.

I'll see how this works.

rbswift, 2021-02-25:
Just about all modern databases allow online backup for the last 10 years. Even if you are somehow using one that can't be backed up online then that would not require mosquito to be restarted along with it. Mosquito can run happily for months and months. Basically for as long as the server is running. HTTP control is fine but uni-directional so quite limited whereas MQTT is pub/sub so much better suited to HA: In fact it's pretty much what it was designed for.

yavinmaster, 2021-03-05:
I use MQTT for several other devices without any issues but for some reason can't get it to reliably work here.  I am however more than happy with how it works by sending curl commands - no issues since changing to this.
I know you don't get the bidirectional interface but you don't really get it using MQTT here either as if anyone uses the standard remote it doesn't result in the position being reported back.

closed 2021-03-05 by yavinmaster