Nickduino/Pi-Somfy #110: Troubleshooting failure of only one shade

nowellag, 2021-03-03:
Starting yesterday one shade stopped responding to scheduled Somfy control.  Have tried removing and adding the shade but am unable to successfully program it into Pi-Somfy again - it won't "jog" when trying to program it.  All the other shades continue to work with both scheduled and manual control via the web interface. The shade responds to the handheld remote. I've tried looking at the operateShutters.log, but not sure what to look for to try and troubleshoot.  Any suggestions?

yavinmaster, 2021-03-05:
I don't know the workings of the rotating code but I had a similar issue to you.  I fixed it by editing the config file and setting the rotating code for that blind to zero (if your rotating code is very high, maybe reduce it by 50 or so), I then kept on trying to operate it from the web interface.  Eventually it started to work, I'm guessing when the rotating code got to where it should be.  I don't know if this happens when the rotating code gets out of sequence somehow.

Someone might say this was a terrible solution - I hope they do, so long as there is a better one.  This only happened to me once but it would be good to have a better solution if it happens again.

nowellag, 2021-03-06:
Interesting suggestion.  Looking at the config file I can see that the other shades have rolling codes around 540, but the problem shade is now in single digits.  I presume this is after trying to newly program it?  I'll try setting it to a number close to , but under the other shade codes and just start trying like you described.  Worth a shot.

Does anyone have any insight as to how the controller obtains the unique shade identifier or the initial correct rolling code?

nowellag, 2021-03-07:
Looks like your suggestion guided me to fixing the problem.  Continuing to use the web interface eventually resulted in it working.  No idea how things got out of sync or how to avoid the issue in the future, but at least there seems to be a resolution!

yavinmaster, 2021-03-07:
Glad it worked for you too.
For me, the issue was associated with trying to get MQTT to work reliably.  Since moving to curl commands and binning MQTT I've not had any issues.  It would be good to know how they get out of sync though.

closed 2021-03-08 by Nickduino