Nickduino/Pi-Somfy #111: Option to reverse direction

dckiller51, 2021-03-11:
hello, thank you for your program. Very useful for my father who does not want a home automation server. Would it be possible to add an option to reverse the direction of the shutter?

Nickduino, 2021-03-11:
You mean with Alexa? The easiest (for us :-D) is to create a routine

dckiller51, 2021-03-11:
> Vous voulez dire avec Alexa? Le plus simple (pour nous :-D) est de créer une routine

No more for manual mode. The idea would be to have during the shutter configuration the choice of inverted the up / down command.

Nickduino, 2021-03-11:
Parce que dans la Marne on fait les choses à l'envers ?

dckiller51, 2021-03-11:
> Parce que dans la Marne on fait les choses à l'envers ?

Mdr. Non mon père à des volets avec deux moteurs pour lequel le sens de monter et descendre est inversé. Cela est arrivée après une coupure d'électricité. Malheureusement les volets ne sont pas indépendant sur le réseau électrique. J'ai essayé les procédures Somfy avec la télécommande sans succès. Donc comme je sais que sur beaucoup de solutions domotique Homeassistant, Jeedom et autre système nous avons la possibilité de cocher une case pour cela, je me dis que cela peut être une solution à intégrer dans votre programme. Après cela n empêche pas le fonctionnement surtout pour les ouvertures et fermetures par le schedule vu que l'on choisi l'action. Donc j'ai créé deux scénarios d' ouverture et fermeture.

Nickduino, 2021-03-11:
Je pense qu'il vaut mieux sur le _scheduler_ ou inverser le sens de rotation des volets que sur l'intégration de ce problème au code.

M'enfin je le note, on sait jamais

closed 2021-03-11 by Nickduino

dckiller51, 2021-03-11:
@Nickduino Je vais essayer la solution du poste #55 
J aurais dû lire avant les questions réponses.