Nickduino/Pi-Somfy #112: rpitx

zbyrek, 2021-03-12:
Did you consider using rpitx? That would reduce hardware to almost none.

Nickduino, 2021-03-12:
Care to elaborate?

zbyrek, 2021-03-12:
Oh, i thought that you are familair with [rpitx](https://github.com/F5OEO/rpitx).
It uses fast GPIO capabilities of raspberry to generate high frequency modulated signal, so basically you could connect just simple wire which acts like antenna to transmit rf (from 5kHz to as high as 1.5Ghz) from raspberry (but low pass filter is highly recommended due to square wave carrier, however for very short transmision periods like in remote for gate it's not such big deal), there is example for OOK modulation, which is used in remote controls.

Nickduino, 2021-03-12:
That's a really smart way of making a transmitter!

But it sounds like you should be the one to fork the project and implement it...

zbyrek, 2021-03-12:
I thought about it but lately i'm lack of time, maybe in near future I found some.