Nickduino/Pi-Somfy #113: Cover only moves by a small amount

Plawasan, 2021-03-20:
First of all huge thanks for creating this, I love being able to solve a 300€ problem for about 3€ ;)

What I have
- 1 shutter in my bedroom
- 3 blinds in my living room
linked to Home Assistant via MQTT. I'm shocked how easy it was to set everything up.

I have been able to pair all my shutters and they all respond to commands. The bedroom shutter works exactly as expected, i.e. the UP command opens it completely, the DOWN command closes it all the way down, my either stops it or sends it to the preset position.

The original 4-channel Telis remote for my living room blinds has both the UP/DOWN buttons and the scroll-wheel for adjusting tilt (+the my button). The My button works as expected however when I send an up/down command from HA to my covers they only move by a small amount, equivalent to one-step move of the scroll wheel.

I've googled, found this https://community.home-assistant.io/t/rflink-somfy-cover-moves-only-a-bit/187904/20 - it is for RFLink however the issue seems the same. The OP there suggest to switch the blinds to US mode, I tried that, no success.

Is there an option to send the "full down/up" command from pi-somfy instead of just the short tilt adjust? Of course being able to do both would be ideal.

Thanks again for the great work.

m.

Plawasan, 2021-04-02:
Update after 2 weeks: I now actually prefer the way it works at the moment (as opposed to ONLY having full up/down). My preset position is blinds all the way down and tilted open - this allows me to reset them using the "stop" command and then tilt them progressively by sending repeated down commands from HA.  It would still be great to have both options available as it's not exactly practical to be closing them fully by repeating the tilt command 6 times (let alone having to open them fully bit by bit - I think the motors would conk out..).

This is a HA script that first sets both blinds to "my" position, then drops both down 3 times to give me partial shade:

https://www.youtube.com/watch?v=wmh4xjTW2ko

Thanks again!

no1knows, 2022-05-26:
Very late to this party but I had this same issue with RFLink. I have six of the (relatively new, I think) Somfy "Lift & Tilt 25" motors, which may be the reason.

There's another thread here (https://www.letscontrolit.com/forum/viewtopic.php?p=58437#p58437) about it where I said "In case useful, I've just discovered that on RFLink32 (https://github.com/schmurtzm/RFLink32-For-Sonoff-RF-Bridge) the up/down commands work properly - i.e. send the blind all the way up/down." I'm not sure why that is - presumably the RFLink32 port uses different code/signals.

If there's any appetite to fix this I'd be happy to test! Cheers.