Nickduino/Pi-Somfy #114: Alexa does not discover device

schweppy, 2021-03-25:
Hello

I have just installed Pi-Somfy on a Pi Zero, as per the instructions. Manual control and scheduling all work as expected, however Alexa does not find my blinds. I have tried discovering devices on the Alexa app and also on various different Echo devices.

Log repeats the following - 

[DEBUG] (Alexa     ) HANDLE REQUEST: b'GET /setup.xml HTTP/1.1\r\nHost: 192.168.0.91:54337\r\nAccept: */*\r\nContent-Type: application/json\r\n\r\n'
[INFO] (Alexa     ) Responding to setup.xml for Living Room Blinds

Any ideas? 

Thanks

Nickduino, 2021-04-01:
Have a look at the closed issues, some of us have encountered problems with Alexa that were solved in the past
I'd say the first thing to try is to reboot your Alexa device

schweppy, 2021-04-01:
I have tried reading all of them unfortunately. Have reset all devices but I have about 6 or 7 different types of Echos and none of them work.

cloudymark, 2021-07-17:
Same here. Tried all the normal. Wonder if the changes to Fauxmo have broken the integration?

ToddGreenfield, 2022-04-21:
Having same issue. Using python3 and starting as a service. Web interface works and no other apparent issues. Alexa just does not discover any devices. Tried with multiple echos and also app on phone.

Nickduino, 2022-05-15:
I created a new blind and Alexa indeed couldn't discover it. But the old ones still work.

vapouryh, 2024-02-11:
https://github.com/Nickduino/Pi-Somfy/pull/163