Nickduino/Pi-Somfy #116: Question: finding shutter names

Beramos, 2021-04-01:
I've been reading through the documentation and the code but I'm wondering how this system knows how many shutters there are since it is only a one way communication through the transmitter?

Nickduino, 2021-04-01:
The system doesn't detect anything, you're right.

**You** will use it to create as many virtual remotes as you need (one per blind seems smart but you can also control several blinds with only one remote) and then register them with the shutters (so the shutters detect the remote, not the other way around)

Beramos, 2021-04-01:
Ok, thanks that makes a lot of sense. Is there more information on which procedure to follow to connect the virtual remotes to the different shutters? Other Somfy projects use the programming button on an existing remote but the installation instructions here do to not explain that part.

correajl, 2021-04-01:
This is my doubt too! I've read about hold the original remote button to enter in some kind of prog state. So, a code from virtual remote is registered. 

After we config Py-Somfy new virtual remote, the original one continues to work? 

Thank you :)

Nickduino, 2021-04-01:
> This is my doubt too! I've read about hold the original remote button to enter in some kind of prog state. So, a code from virtual remote is registered.

Yes. Enter program mode on the blind with the original remote (or powering the shutter on and off for a set amount of time or whatever the procedure for your motor is), then press the program button on the new virtual remote, so that the blind register its address too.

> After we config Py-Somfy new virtual remote, the original one continues to work?

Yes. You can have a few remotes acting on one blind.

Beramos, 2021-04-01:
Thanks, I'm assuming you should enter the programming mode on the
remotes/shutters before the first time running operateshutter.py?

Nickduino, 2021-04-01:
No need. You can use the interface to create all your virtual remotes today and register them in a week, it doesn't matter. What does matter is that you place your shutter in programming mode before you hit "program" on the virtual remote (on the Add/Remove Shutter tab of the webapp) .

**If** you want to add several shutters to one virtual remote, do them one by one:
1. Long press (until the shutter moves up and down a bit) program on physical remote 1
2. Press program on your virtual remote
3. Long press (until the shutter moves up and down a bit) program on physical remote 2
4. Press program on your virtual remote
  => Both your shutters will move up or down when you send a signal with your virtual remote. Your physical remotes will stay attached to their respective shutters.

Beramos, 2021-04-01:
Ok thanks for taking the time to explain this. I was not aware that the use
of the webapp was necessary for the setup. I'll order some transmitters and
try to get this beauty working!

Nickduino, 2021-04-01:
> I was not aware that the use of the webapp was necessary for the setup

It's not a necessity but it's much easier.

closed 2021-04-01 by Nickduino