Nickduino/Pi-Somfy #118: Stuck On Entering MQTT Polling Loop

Ben10lightningx, 2021-06-02:
I believe that I have configured everything with mqtt correctly, however, in the log file it just says entering mqtt polling loop and nothing shows up
 in homeassistant. How do I fix this?

[operateShutters.log](https://github.com/Nickduino/Pi-Somfy/files/6581522/operateShutters.log)

incub77, 2021-06-21:
Please try updating to the latest version. I had a similar issue, which was fixed with [https://github.com/Nickduino/Pi-Somfy/pull/98](https://github.com/Nickduino/Pi-Somfy/pull/98).

Ben10lightningx, 2021-08-22:
What do you mean by updating to the current master? I replaced the mymqtt.py file with the newest version of it and I am still having the same issue.

closed 2021-09-11 by Ben10lightningx