Nickduino/Pi-Somfy #124: just a question: How to register remote controls via Pi-Somfy

W11T, 2021-09-04:
Sorry, i couldn't find a forum. Is it ok, to aks questions here?

My Somfy Telis 4 RTS are messed up. Pi-Somfy is working well. 
Is it possible to reregister the remote controls using Pi-Somfy?

rbswift, 2021-09-04:
I don't think Pi-Somfy supports long-press which is needed to add a remote. You might be able to change the code slightly to send more repeats and get the motor to respond appropriately but it's probably easier to just reset the motor by power cycling it off 2s, on 10s, off 2s, on

Nickduino, 2021-09-04:
Agreed with @rbswift

W11T, 2021-09-04:
> it's probably easier to just reset the motor

Unfortenately they are papered over 

> You might be able to change the code slightly to send more repeats and get the motor to respond appropriately

Could somebody push me in the right direction?
Any help appreciated.

rbswift, 2021-09-05:
> Unfortunately they are papered over

You can use the breaker to power cycle. Doesn't need to be done locally

> Could somebody push me in the right direction?

Try increasing the 1 in line 207 of `operateshutters.py`

`self.sendCommand(shutterId, self.buttonProg, 1)`

W11T, 2021-09-26:
I'm sorry for answering late.
repetition = 16 did the job! I was able to fix all of my remote controls!

Thanks alot, Nickduino and rbswift, for the good App and for the good Support!

closed 2021-09-26 by W11T

bobf41, 2021-09-27:
> I'm sorry for answering late.
> repetition = 16 did the job! I was able to fix all of my remote controls!
> 
> Thanks alot, Nickduino and rbswift, for the good App and for the good Support!

That repetition is that about 
self.sendCommand(shutterId, self.buttonProg, 16) or another line?

W11T, 2021-09-28:
> That repetition is that about
> self.sendCommand(shutterId, self.buttonProg, 16) or another line?

Yes, correct.