Nickduino/Pi-Somfy #125: One Specific Shade Not Working

Ben10lightningx, 2021-09-11:
I have multiple shades that are working just fine with pi somfy. One of these shades had been working fine for 1 year now until a few months ago it just randomly stopped responding to all commands. I have tried re adding it. I have also tried removing all mention of this one shade from all of the files and it still does not work. This shade is also one of the closest ones to the pi. All of the telis remotes are still able to control it just fine.

lepirlouit, 2021-09-15:
is it possible that the Shade is to far away of the emitter or reception is obstructed ?

Ben10lightningx, 2021-09-19:
No because it is the second closed one to the pi and there are no walls between them.

rbswift, 2022-04-23:
Is it a battery powered motor? I've noticed these sometimes need a few more repeats to get them moving

kvuurens, 2022-07-13:
I am having the same issue. When I click "Try programming again", I see the shutters jog, but I cannot control the blinds.

MichaelB2018, 2026-03-26:
This was most likely caused by the upper-/lowercase hex ID bug (see #166), which created duplicate entries with wrong rolling codes in the `[ShutterRollingCodes]` config section. This has been fixed in release 3.0 ([PR #170](https://github.com/Nickduino/Pi-Somfy/pull/170)). Before updating, please manually clean up your `operateShutters.conf` file — look for duplicate shutter addresses in `[ShutterRollingCodes]` that differ only in case (e.g. `0x27962a` vs `0x27962A`) and remove the lowercase duplicate, keeping the entry with the correct rolling code. After updating to 3.0, the issue should not recur. Please reopen if the problem

closed 2026-03-26 by MichaelB2018