Nickduino/Pi-Somfy #127: Program remote for Somfy Meteolis 6 (RTS)

bobf41, 2021-09-22:
I have a Somfy Meteolis 6 in my house, I placed the Raspberry Pi 3+ next to it with the transmittor connected to the defaults pins and the right Oscilator.

On the Meteolis I can add a new remote where the Meteolis waits for a short push on the program button of a remote.

Here is where is stops, when I press the program button in the WebUI of PI-Somfy nothing is received.
I run Python 3.7.3 as default.
sudo python /home/pi/Pi-Somfy/operateShutters.py runs fine without errors or warnings.

The Meteolis is RTS and is being controlled by the normal Somfy Telis RTS remotes.

Any ideas here?

Nickduino, 2021-09-22:
I don't know the Meteolis but it seems to be some sort of central remote,
correct?
In that case, I'm not sure what you wanna do...please write a few
sentences on how you wanna use Pi-Somfy in your system

bobf41, 2021-09-22:
The Metolis is the central system which controlls the wired motors.
It receives the RTS remote signals and lowers or highers one of the connected motors.
So basically instead of RTS enabled shutters there is a central RTS receiver which then controls the wired motors.

I would like to be able to send these Somfy RTS signals with the webinterface and finally with HA but I believe your webinterface should work first ;-)

Nickduino, 2021-09-22:
Alright.
Then it's the same kind of issues people encounter when they want to pair a
shutter. Try having a look at past issues. I would start by making sure the
Meteolis can register a new remote, check the wiring 100 times, check the
emitter/crystal, re-check the wiring, possibly unregister one of my remotes
and try to register Pi-Somfy's remote.

Good luck!

bobf41, 2021-09-22:
Ok, I did check registering the normal remote again, this is fine, press program button on the back and it is picked up already.

The crystal is properly installed and the 3 cables are connected right to the pins, I understand there is no way to monitor if the transmitter is even noticed by the Pi right?

Pressing the Program button under action is basically the program button of a normal remote right?
[image]

Nothing to check in the config file.
TXGPIO = 4

Thanks for your quick answers.

Nickduino, 2021-09-22:
I would still rerecheck the wiring.
Possibly the crystal and emitter ? Replacing that crystal is not very easy
operation, sometimes we damage the emitter (or the emitter didn't work in
the first place)

Monitoring: the Pi doesn't "notice" the emitter, it just sends a signal and
it would still send it if the emitter wasn't there. If you have a logic
analyser, you can see what data is sent to the TX pin (and possibly the
antenna). You could check the emitter work by sending anything through the
air to a receiver (possibly connected to the Pi, preferably connected to
something else like an Arduino or another Pi). You could check the emitter
AND Pi-Somfy with an SDR or something a Ham Radio enthusiast uses (not my
area of expertise)

Pressing the program button: yeah, that's similar. The only thing is our
program button sends a set amount of frames while the button of the remote
keeps sending it until you let go of it. You could find that number of
repetitions in the code and try increasing (or, more surprisingly,
decreasing) it. Please report back if that's it, that would be interesting.

Nickduino, 2021-09-22:
> You could check the emitter AND Pi-Somfy with an SDR or something a Ham Radio enthusiast uses (not my area of expertise)

There: https://github.com/Nickduino/Pi-Somfy/issues/49

bobf41, 2021-09-25:
Just attached the AMP meter, and I see current flowing when cammands are send, when I pull the data cable this stops so wires are good. Even tried an antenna connected with no succes.

So I received wrong crystals, when crystal has failed or is wrongly installed this current wouldn't flow right?

Nickduino, 2021-09-25:
You attached it to what? The antenna ?

I can't answer that with any certainty :-(

Check the number of repetitions in the code

bobf41, 2021-09-25:
I put a multimeter in series of ground so can measure the current which is flowing in mA. 
Just created a secondary one with the R433.4 Crystal on it and gives the same result. (No result actually) 😞

Nickduino, 2021-09-27:
> You attached it to what? The antenna ?

You haven't answered that one

> Check the number of repetitions in the code

https://github.com/Nickduino/Pi-Somfy/issues/124#issuecomment-913077179

bobf41, 2021-09-27:
Attached it to a multimeter. 

Tried repetition of 1-5 and 16 written in the other post and restarted Pi-Somfy in between

ngutman, 2022-02-10:
I had a similar issue where I could not get my virtual Pi-Somfy remote to register using cc1101 transceiver (had to modify the code a bit). After hours of debugging it was related to the "encryption key" (see reference [here](https://pushstack.wordpress.com/somfy-rts-protocol/)) - the first byte in the sent frame, it's hardcoded as [0xA7](https://github.com/Nickduino/Pi-Somfy/blob/master/operateShutters.py#L255), but it seems it does have a meaning :)

After decoding frames from my original remote, and comparing them to what was sent - I noticed the first byte was different, once I modified the original value and changed it from 0xA7 to match the one I recorded everything worked.

Here's the recorded signal (using [urh](https://github.com/jopohl/urh)) -

[image]

```0110011010010110101001100101011010100110011001100110101001011010100110010110010110010101011001100101010101010101```

Which decodes to 
```
Frame = 59d1d573a48500 (length = 7, bin=01011001110100011101010101110011101001001000010100000000)
De-xored Frame = 598804a6d72185
First Byte = 0x59
Rolling Code = 1190
Button = 0x8
Address = 0xd72185
```

🤷 
Hope this helps someone

Nickduino, 2022-02-11:
> Here's the recorded signal (using [urh](https://github.com/jopohl/urh)) -
> 
> [image]
> 

That's cool!

Just to be clear: you changed the encryption key but kept a different remote address than the one of your original remote?

ngutman, 2022-02-11:
@Nickduino yup, exactly

bobf41, 2022-02-12:
Tried Thijs code directly but without succes. Perhaps it is some other code at my side off course. Or the meteolis just binds differently although normal RTS remotes can be programmed easily. On screen add transmitter for channel x. Press program shortly and ready.

ngutman, 2022-02-12:
@bobf41 you can try to iterate on all first 4 bits of the encryption key byte (16 values - 0x0-0xF). But I think if that doesn’t work the next best thing you can do is try and record the original remote.