Nickduino/Pi-Somfy #129: Flawed crystal or poor soldering or should this have worked?

sfla, 2021-10-09:
i just can't get it to work. I bought a 5-pack of transmitter and crystals, and have tried soldering it twice now. And nothing happens. 
As suggested (by @Ezezi3l ) in another thread I got one of these SDR-receivers and tried looking at what's actually being transmitted:

[image]
The left red line is 433.42MHz, and the "wide" signal (bottom left) is when I pressed the physical existing remote for a second. So the existing remotes are definitely on 433.42MHz.
The dots on the right side is what appears when I click the button on the Pi (three times). Apparently, my signal is transmitted at about 433,5625MHz.

What can cause this? Can bad solder shift the frequency this much? Is the crystal bad or damaged? Is there anything I can do? Or is it close enough that it should be working?

sfla, 2021-10-09:
I found my previous soldering attempt and plugged it in to analyze the signal, and it turns out that one never sent anything.
But I soldered a third attempt now, and the exact same thing happens. 
The signal is exactly 433.55 with a new transmitter and a new crystal. (Same as earlier, even though the post/picture says 433.5625. I was apparently hovering the cursor a bit off).

Maybe the entire batch of crystals is flawed. It physically says "433.4" on the crystals, and they sold them as "433.42", so im pretty sure they're the correct ones.
I hope there's another solution than to buy new crystals, because it literally took more than 3 months having them sent here from China.. 

Anyway, I have three different physical remotes for three different sets of blinders. Two of them transmit on exactly 433.42, the third one actually transmits at exactly 433.45. Consistently. This suggests to me that there is some wiggle-room regarding frequency here, because I'm pretty sure these blinders are identical and installed at the same time (before I moved in).

So if it's not the frequency, can it be something about the transmitted data? I do notice that the signal from the original remotes are visually quite different from the pi-signals, in this SDR-visualizer thing. I have no idea how to use this SDR-scanner/reader/whatever properly, so I don't know how to inspect the payloads, but maybe there's something there?

MarkGodwin, 2021-10-30:
My SDR arrived, as well as two new transmitter modules, so I did some investigation...
* The real Somfy remote is transmitting at 433.**42**.
* My bodged transmitter with the capacitor installed is broadcasting at 433.**24**Mhz. Again, this seems to work fine.
* The second transmitter, unmodified, broadcast at 433.92 - no response from the blinds.
* When I changed the crystal, it broadcast at 433.**42** with no aerial connected, and it dropped a tiny bit when I added the aerial.
* Adding a capacitor dropped the frequency to 433.35. It continued to operate the blinds.
* The last transmitter, unmodified, was all the way up at 434.0Mhz. With the crystal replaced, and aerial attached,  it broadcasts at exactly 433.42! This is the one I am going to use!

My original transmitter that has never worked doesn't appear to transmit any signals at all, so I must have broken one of the components - or it was faulty in the first place.

Other points I would make...
The signal strength of the transmitter seems to pretty much match that of the Somfy remote - in that it "sounds" about the same, and the spectrum analyser looks about the same. This seems to be the case regardless of whether there is an aerial connected.
I can't see any difference in the signal from my "good" transmitter to that of the Somfy remote in the spectrum analyser, unlike your trace above. Therefore, I suggest there _is_ something wrong with your transmitter, either in the wiring, the soldering or the crystals. 

I'm kind of tempted to smash the capacitor off my previous transmitter to see what it broadcasts at then (if at all). My guess is it's not oscillating properly until the capacitor is attached.

MarkGodwin, 2021-10-30:
I added a capacitor to the completely broken one, but when testing the continuity found that the little surface mount inductor (the thing underneath the missing second small coil) wasn't connected. After dabbing some solder on that, it now transmits a little bit above 433.42 _(with capacitor)._

Nickduino, 2021-11-01:
> Maybe it will give you a chuckle.

It absolutely did

> I'm kind of tempted to smash the capacitor off my previous transmitter to see what it broadcasts at then (if at all). My guess is it's not oscillating properly until the capacitor is attached.

We **need** to know!

Scrath1, 2022-04-24:
I also have a lot of problems getting the transmitter to work properly. I had one functioning last year that somehow stopped working over the winter. With no way to troubleshoot it I also ordered and SDR and was able to see that it didn't send any signals anymore.
Luckily I still had some leftover transmitters and crystals from the previous attempt so I first tested an unmodified transmitter which worked flawlessly at 433.92Mhz. Then after desoldering the crystal I wanted to try multiple of my remaining 433.42Mhz crystals to see if a some were more accurate than others (based on what I read here).

To try that out I soldered some wires to where the crystal would be attached and plugged them into the breadboard so I could more easily swap crystals. Then nothing. It didn't work. I was already hovering above the buy button for a new bunch of transmittors when I decided I would first try properly attaching the crystal how its meant to be. Suddenly the transmitter worked again although it was pretty weak without an antenna and wasn't transmitting on quite the correct frequency.

Nevertheless I resoldered it onto a tiny perfboard so I can directly plug it into my Pi 1 which I am using as a server for this. The first test without an antenna worked and I was able to move the shutter again.

Next I attached my homemade antenna which I had successfully used with my previous attempt last year. Suddenly the transmitter stopped working again.

I have an arduino set up on my desk where I can plug in the transmitter which will send a signal once per second. After I couldn't get the transmitter to work in the Pi I plugged it into the arduino again where it worked flawlessly.

Thinking that it worked perfectly before attaching the antenna I desoldered it again but still couldn't get it to work. Tried it again using the arduino. It sends a signal without any problems.

So what is the difference between the arduino and the Pi? The first thing I though was that maybe the Pi was shielding the antenna or something but I have no background in all this radio stuff so I have no clue.

The other difference is that the arduino is connected to the antenna using jumper cables while the adapter board enables me to plug it into the Pi directly. So the next thing I tried was using some dupont jumper cables to connect the transmitter to the Pi (still without an antenna attached). Suddenly its working again.

Next I reconnected the antenna while still using dupont cables for the connection between the Pi and the transmitter and it is still working (and sending on a frequency of 433.46Mhz).

To test if the positioning on top of the Pi caused some issues I held the transmitter above the Pi and let the program sent a stop command. The strength of this signal was unchanged no matter how I positioned the transmitter on top of the Pi or next to it.

This leads me to believe that the length of the wires going to the transmitter are the cause of it working or not working. I will try using a bigger antenna next without the dupont wires

For the record, this is how my transmitter currently looks (using the homemade antenna that I had success with last year). I have the crystal standing up a bit because I read in some issue that this fixed someones problems though I don't know if this actually helps at all.
[image]
[image]

Scrath1, 2022-04-24:
After attaching a longer antenna stolen from a CC1101 I still have the same results. With dupont cables between the transmitter and the Pi it works. Without it doesn't.

Here's a picture of the antenna I used for this test
[image]

Anyway, another thing I noticed during testing was that my Pi is continuously broadcasting *something* as soon as I plug in the USB cable before I send the first command.

This is what the signals look like while booting the Pi
[image]

and here's what it looks like as soon as I send a first signal
[image]

Maybe it would make sense to manually set the data pin to low when starting the program?

MarkGodwin, 2022-04-24:
I knew it before, and I still know it now - the dark art of analogue radio circuits is well beyond me. However, mine has been working reliably since my random escapade with capacitor soldering. I _am_ using dupont jumper cables, though. I also have a little mount inside my project box which keeps the transmitter a little way away from the PI.
What kind of PI are you using? I'm on a Model 2b. Perhaps there is some interference from USB3 or something like that on newer models?

Scrath1, 2022-04-24:
I am using an ancient Pi 1 since its so underpowered that I don't know any other projects for it. It's basically dedicated just to this project. Definetely no interference from USB3 though ;D

I had also tried using a capacitor when attempting to fix my old previously working sensor. What capacity did you use? The smallest I have is 0.22uF. I might try that next. I had mine soldered on the 2 empty pads on the side where the antenna is connected. Is that the correct spot?

MarkGodwin, 2022-04-24:
I pinched this one off the transmitter board. I'm not sure it really helped at all, but it did appear to alter the frequency, and did magically restore a non-working transmitter
[image]
.

Nickduino, 2022-04-26:
No idea if that helps but my transmitter is on top of my Pi case WITH A GROUNDED SHIELD IN BETWEEN and long-ish cables.

Scrath1, 2022-04-30:
So I finally had some time to experiment some more and found out that the antenna length and shielding from the Pi are irrelevant. Also the cables don't matter either.

To test this I first started to use an antenna that I received together with my SDR that is specifally meant for 433MHz and tried it without any luck.

Next I tried to put some shielding between the Pi and the transmitter by using a grounded piece of aluminum foil that was isolated using duct tape. This also didn't help at all but at least it looked kind of funny in my opinion.
[image]

After nothing else had worked I decided to test all connections once more using the multimeter which was the moment I nearly facepalmed myself.
[image]
I had a short between the data and a ground pin when the transmitter is mounted. The reason I hadn't detected this earlier was that this ground pin wasn't meant to be used at all so it is only connected when the transmitter is directly mounted to the Pi.

On the upside I can now confirm that the frequency for the blinds doesn't have to be exact. My transmitter is sending at roughly 433.5MHz and my blinds still react to it. I can also confirm that touching the oscillator with my finger changes the frequency toward the desired 433.42MHz though it doesn't reach it.

I can also quite confidently say now that the antenna length doesn't matter beyond the recommended 17cm since I tested antennas that were much longer. I can't say if it will be the same with shorter antennas though.

 I didn't manage to test the capacitor that MarkGodwin recommended since I couldn't get one off the receiver board. Instead I tested a 0.22uF capacitor which caused the transmitter to not function at all.

Scrath1, 2022-06-29:
Small update from me: The transmitter stopped working again. Apparently the oscillator died because replacing it fixed the issue. I wish I knew why they keep dying though

Nickduino, 2022-07-01:
Glad it's working :-)

> On the upside I can now confirm that the frequency for the blinds doesn't have to be exact. My transmitter is sending at roughly 433.5MHz and my blinds still react to it.

0.08 MHz is pretty close in my opinion. I wouldn't work with 433.9, though.

Agreed on the antenna: the blinds aren't too picky (probably because they're tuned to 433.42 MHz), mine kept working with a broken antenna...

closed 2022-07-01 by Nickduino

Nickduino, 2022-07-04:
> So I factory reset the two misbehaving blinds. (https://www.youtube.com/watch?v=sTGraGEfEH4) I then had to train the original Somfy transmitter again. And then I was able to add the Pi controller to the blinds with no problems. (https://www.youtube.com/watch?v=z3FQa_RRXKw)

GREAT!

I THINK I've read around 10 remotes but that may vary from blind to blind