Nickduino/Pi-Somfy #134: Newer `requirements.txt` versions incompatible with pi-zero

janstenpickle, 2022-01-02:
As per title, I'm not sure which requirement is causing the problem, but I suspect it's `ephem`.

When trying to run `operateShutters.py` all you get is an `Illegal Operation` message, probably because a newer version of one of the dependencies has native code built for a newer instruction set.

I've found pinning the versions in `requirements.txt` fixes this, I suggest that either the versions are pinned in this repo or an alternative `requirements.pizero.txt` be added containing something like this (my last known good versions):

```
ephem==4.0.0.2
configparser==5.0.2
Flask==2.0.1
paho-mqtt==1.5.1
pigpio==1.78
```

Nickduino, 2022-01-02:
Care to do a pull request? :-)

janstenpickle, 2022-01-02:
@Nickduino sure thing, would you prefer a modification to the existing `requirements.txt` or a new one for Pi-Zero's?

Nickduino, 2022-01-02:
> would you prefer a modification to the existing `requirements.txt` or a new one for Pi-Zero's?

@MichaelB2018, @rbswift, @MarkGodwin and others: what's your take on this?

MichaelB2018, 2022-01-02:
suggest a new one

rbswift, 2022-01-03:
I'm not an expert but suggest holding back versions in as targeted and limited was possible so we get maximum benefit of future but fixes where possible.

Creating a virtual environment (venv) may also be an option.

Nickduino, 2022-01-03:
Alright, a limiting requirements.txt for the Pi-Zero, then :-)