Nickduino/Pi-Somfy #136: PIGPIO timing off by a factor of 2

harebrainedkiwi, 2022-02-15:
Problem: Nothing happened when trying to add a new shutter (the shutter jogs with a 2s press of the programming button on the remote, but then the Pi-Somfy transmission was ignored by the shutter).

Using a 433MHz receiver and oscilloscope, the frame timing from Pi-Somfy is twice as long as that from the official remote.  E.g. the hardware sync timing described at [pushstack](https://pushstack.wordpress.com/somfy-rts-protocol/) is 2416us high/low, but the measured timing from the Pi is 5120us high/low.  This appears to be a problem with PIGPIO, as a quick search found a similar discussion on the topic [here](https://github.com/joan2937/pigpio/issues/331).

As a hack, modifying all the timing within the code makes the shutters work correctly. e.g. change from
`wf.append(pigpio.pulse(1<<self.TXGPIO, 0, 640)) `
to 
`wf.append(pigpio.pulse(1<<self.TXGPIO, 0, 320))`

As this is not an issue within the Pi-Somfy code itself, I'm not sure what the best approach is? Some options might be:
1. Add this info to a known issues section of the readme.md?
2. Since the transmission length is a known fixed duration, add a timer within the code to measure the transmission time (similar to [here](https://github.com/joan2937/pigpio/issues/331#issuecomment-598586623)).  This measured time can then be checked and if it exceeds a threshold report the 2x error to the user?

Some additional debug info:
pigpio V79

`cat /proc/cpuinfo`
Hardware        : BCM2835
Revision        : a02082
Model           : Raspberry Pi 3 Model B Rev 1.2

Nickduino, 2022-02-16:
You're running the current v79 release?

Did you run [the test](https://github.com/joan2937/pigpio/issues/331#issuecomment-598925717) and confirmed your clock had a factor of 2?

Have you considered switching your pigpio's clock to PWM?

harebrainedkiwi, 2022-02-16:
> You're running the current v79 release?

Yes, v79 (as mentioned at the end of my previous comment).

> Did you run [the test](https://github.com/joan2937/pigpio/issues/331#issuecomment-598925717) and confirmed your clock had a factor of 2?

I hadn't gotten that far with debugging yesterday (I was so happy that my shutters were finally working and I then moved on to configuring MQTT+HA!).  It makes sense to spend a bit more time to help others that might also run into this problem, so I just tried pulling [hayguen's repo](https://github.com/hayguen/pigpio) and running sudo make test, and it confirmed the PCM clock ratio is 2x.  The good news is that the PWM clock appears correct:
```
testing PWM clock
  measured clock ratio = 1.004
  PASS: clock is OK
testing PCM clock
  measured clock ratio = 2.008
  FAIL: clock is skewed!
```
> Have you considered switching your pigpio's clock to PWM?

I just tried running the code below (using the unmodified operateShutters.py file with the original timing) and it has fixed the problem :-)

```
sudo systemctl stop shutters.service
sudo killall pigpiod
sudo pigpiod -t0
sudo systemctl start shutters.service
```

marek-polak, 2022-10-17:
I had the same problem with my setup recently as @harebrainedkiwi mentioned. 
After several hours troubleshooting what may be wrong with my hardware and several re-soldering attempts later, I've found this thread and checked the PCM clock in the Raspberry Pi...and oh boy, it was 2x off. 
After switching pigpio clock to PWM, it suddenly works as charm! 

I think it should be worth putting @harebrainedkiwi solution in the project readme, so it would be visible to others (or creating a dedicated troubleshooting.md file).

Anyway, thank you @Nickduino for the awesome project and @harebrainedkiwi for detailed problem description & solution.

My Raspberry Pi model:
```
Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000fc49f874
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
```

Nickduino, 2022-10-17:
> I think it should be worth putting @harebrainedkiwi solution in the project readme, so it would be visible to others (or creating a dedicated troubleshooting.md file).

We welcome pull requests :-)