Nickduino/Pi-Somfy #137: Multiple channels activated on press

pedroandrade1977, 2022-04-01:
Hi, first of all i wanted to thank you for this great work. It is a great solution to integrate my pergola with my automation platform.

While I managed to initially register 4 channels of my remote (control pergola roof, vertical shade, light 1 and light 2), after a while it stopped working and in trying to find the cause i wanted to delete the entries and start from scratch. I tried pressing delete in the web app, but everytime i reloaded the page the entries showed up again. So, I ended up deleting the entries from the conf file manually. Btw the problem was hardware but I didn’t realize before deleting the entries.

the problem now is that, after adding the channels again from scratch and doing the programming process, the actions are mixed, and one button is activating more than one action, while others are not working at all. The physical remote continues working normally.

Did the motors get “confused”? Is there some way for the motors to forget the PI and start from a clean slate? Or how else could I go about trying to fix this?

Your help is much appreciated

Nickduino, 2022-04-01:
> Is there some way for the motors to forget the PI and start from a clean slate? Or how else could I go about trying to fix this?

Yes, there is always a way to empty their memory check the documentation of your motors. On mine, I have to disconnect them from mains power for 2 seconds, reconnect them 7 seconds, re-disconnect them 2 seconds and reconnect them.
That way, they will forget **all** remotes and you'll have to repair your original remote too.

Now, I'm not sure you need to do that (pairing the physical remote is a bit daunting to most people). Why don't you start over with four new virtual remotes? But this time, make sure you pair them one by one to the right blind.

Alternatively, you could force your pergolas to forget only the virtual remote: enter the pairing mode with the original physical remote, then press the program button of the virtual remote you want the pergola to forget.

pedroandrade1977, 2022-04-01:
“ Alternatively, you could force your pergolas to forget only the virtual remote: enter the pairing mode with the original physical remote, then press the program button of the virtual remote you want the pergola to forget.”

I will definitely try this. But can it forget the ones i deleted by hand from the config file? Or after deleting it just recreates the “same” virtual remote?

Nickduino, 2022-04-01:
I don't think that's an issue: either they're the same and your pergola will forget them, or they're not and you don't care if your pergola remembers the address of a virtual remote you don't have or will ever have access to.

pedroandrade1977, 2022-04-02:
Hi thanks for your help. I did have to reset the lighting receiver and re-register both physical and virtual remotes. For the two motors this was not needed. For now it is working fine.

Just a couple of issues I find during creating/removing remotes:
1. Everytime I add a remote, the web app gets stuck on a loading icon, no matter if i reload the page. The only way i found to recover is to restart the service with systemctl. The remote is saved though.
2. If I try to delete a remote, it initially disappears from the web app, but upon reload it’s still there. The only way I managed to delete is to erase from the conf file.

In any case, many thanks for your work on this, very happy with the outcome.

note: i am using a Pi Zero 2 W

Nickduino, 2022-04-02:
Glad it's working.

Hopefully someone else (@MichaelB2018 ?) will have a look at the interface since it's not my forte at all

malys, 2023-06-16:
Hi,
I solved my problem.
There is a bug from configuration manager: shutterid case is not consistent #159 
```
 it could be possible that the origin of problem is the reuse of shutter I'd 0x....?
```
creating ghost shutters to not reuse the same id, I haven't any problem. 
Thanks @mihaiaka for your answer.

billmoseley, 2023-06-28:
I have what appears to be a similar issue. I have six Somfy blinds (5 on one remote, and 1 extra remote). I'm using a RPi3B and the FS100A with swapped oscillator.

When I raise or lower "Living Room Left" the "Nook Left" will also move. But, moving "Nook Left" does NOT move the "Living Room Left".

If I delete both of them and then set them up again new the correct blind jogs for each -- although I assume that is because I pushed the program button on the remote for only one blind.

Is it that the RF signal is not "clean" enough and both blinds are hearing their own code? I would have thought the rolling codes would prevent that. The rolling codes are close, if that matters:

```
[Shutters]
0x279621 = Living Room Left,True,11
0x279622 = Living Room Right,True,11
0x279623 = Nook Left,True,8
0x279624 = Nook Center,True,10
0x279625 = Nook Right1,True,8
0x279626 = Kitchen,True,8

[ShutterRollingCodes]
0x279621 = 7
0x279622 = 27
0x279623 = 9
0x279624 = 17
0x279625 = 15
0x279626 = 19

```

What can I do to debug further?