Nickduino/Pi-Somfy #138: Can this be used concurrently with a normal somfy rts remote?

Scrath1, 2022-04-17:
Basically the title.

Can I use this while still using a normal somfy remote from time to time?
As far as I understand this project has no rf receiver so I figure when I use a normal remote the rolling codes would go out of sync and this would stop working?

Nickduino, 2022-04-17:
> Can I use this while still using a normal somfy remote from time to time?

Yep

> As far as I understand this project has no rf receiver so I figure when I use a normal remote the rolling codes would go out of sync and this would stop working?

Nope since the project will create one (or several) virtual remotes with its own address and rolling code

Scrath1, 2022-04-17:
Thank you. I guess I misunderstood the protocol. This makes more sense

closed 2022-04-17 by Scrath1