Nickduino/Pi-Somfy #142: Somfy Pi

3DGenius79, 2022-05-21:
Hallo zusammen

Ich kann alles programmieren aber Alexa findet keine neuen Geräte, was habe ich falsch gemacht?

Danke

Nickduino, 2022-05-21:
> was habe ich falsch gemacht?

I don't know, maybe nothing. Have you tried turning your Alexa off and on again?

Others have reported that same issue (not being discoverable by Alexa) but I don't know what's causing it. Maybe you should have a look at the fauxmo documentation.

3DGenius79, 2022-05-22:
Hi
Hab ein neustart gemacht, ohne erfolg.
Habe alle nochmal Installiert und von punkt 3 Software alle befehle installiert. sudo python3 /home/pi/Pi-Somfy/operateShutters.py musste ich im Ordner Pi-Somfy installieren sonst ging es nicht.
Punkt 4 Befehls liste habe ich bekommen. Dann ging ich auf Punkt 5 Web-Gui und habe meinen Shutter hinzugefügt mit erfolg. Nachdem wollte ich Alexa fragen ob Sie neue Geräte gefunden hat, ohne erfolg. Muss ich die Befehle bei Punkt 4- 3-6 auch hinzufügen? Hatte ich das erste mal auch gemacht aber diesmal nicht.

Danke

3DGenius79, 2022-05-22:
Muss ich vielleicht ein Skill aktivieren?

Nickduino, 2022-05-23:
> Muss ich vielleicht ein Skill aktivieren?

You shouldn't have to, Alexa should discover the WeMo devices without the skill. It won't hurt, though

3DGenius79, 2022-05-23:
Sorry

Hi Have made a reboot, without success.  Have all again installed and in point 3 software all commands installed.  sudo python3 /home/pi/Pi-Somfy/operateShutters.py I had in the pi Somfy folder does not install otherwise it went.  Item 4 command list got me.  Then I went to Point 5 Web-Gui and have my shutter added with success.  Once I wanted to Alexa ask whether you have new devices are found, without success.  Do I need to add the commands at point 4 3-6 too?  I had also made it the first time but not this time.

 Thanks

Nickduino, 2022-05-23:
I don't know what you mean by adding the commands at point 4 3-6.

I still think you should have a look at the fauxmo documentation. Maybe something has changed either with fauxmo or with Alexa.

Maybe use a packet sniffer to see what happens (if Alexa does ask devices on the network to identify themselves and if Pi-Somfy answers). This is far outside my comfort zone, though, so you'll have to find help elsewhere if needed.

3DGenius79, 2022-05-24:
Thank you for you answer.
I look what I can do

closed 2024-01-14 by Nickduino