Nickduino/Pi-Somfy #144: Infinite loading time

Assassinee, 2022-06-19:
Hello all,

I have a problem controlling my shutters.

I have a Raspberry Pi 1 B and a Raspberry Pi 4B. First I tried it on the 1 and now on the 4.

I should not have made a mistake with the soldering. I also checked the connections several times and they are plugged in correctly.

During the installation there were no problems and everything worked fine.

I start the webinterface with the following command
`sudo python3 /home/pi/Pi-Somfy/operateShutters.py -c /home/pi/Pi-Somfy/operateShutters.conf -a `

First I select a location. Then during the teach-in the problem appears.
I press the programming button on the remote control. The shutter moves up and down for a short time. Then I click on the field it worked. Thereupon the web page is only loading again and nothing happens.

In the config file, however, an ID for the shutters was entered. 

Translated with www.DeepL.com/Translator (free version)

huanvu, 2022-06-25:
@Assassinee most likely fixed by https://github.com/Nickduino/Pi-Somfy/pull/145

Assassinee, 2022-06-26:
@huanvu Thank you for your help. Now I can control the shutters with the Pi 4B.

Unfortunately, I still can not manage to control the shutters with the Pi 1B. It tells me now that it sends the command, but nothing happens. Can you possibly help me there too?