Nickduino/Pi-Somfy #151: Can´t pair my Shutters

NoI3ody, 2022-12-16:
Hello,

i try to pair my shutters with the Somfy-Pi but they are not responding to it.
I have two Somfy Telis 4 RTS Pure Remotes with Number 74300 printed on them.

I made an Somfy-Pi first with FS1000 Emitter and now with CC1101 Emitter.
With the FS1000 i can´t get the frequency to match, with the CC1101 i am now between my two telis remotes.

When i start the programming of my blind i press on one of the remotes the programming button and afterwards the shutter jogs up/down. Pressing the programming button in Somfy-Pi does not do anything.

I looked now with my SDR what the remotes are sending if i press the middle button and when i press the middle button in somfy-pi and the results are not really the same?! Do i have the wrong remotes because it says RTS Pure on the back?
[image]

Nickduino, 2022-12-24:
> I looked now with my SDR what the remotes are sending if i press the middle button and when i press the middle button in somfy-pi and the results are not really the same?!

What does that mean? They're both sending something at the right frequency, right?

But that something changes every time, the protocol uses a rolling code.

NoI3ody, 2022-12-25:
Yes, they both send something at the right frequenzy.

But on my Telis it seems to be a single burst all the time and the pi sends multiple bursts as you see in my screenshot.
I think that could maybe be the problem why i can´t sync the shutter with my pi?

Nickduino, 2022-12-25:
The script simulates a slightly longer press, I don't see how that works be an issue.

Nickduino, 2022-12-26:
Check your PiGPIO timing: https://github.com/Nickduino/Pi-Somfy/issues/136#issuecomment-1041907959

NoI3ody, 2023-01-15:
Thanks fot the tipp with the timing.
My PCM-Clock is off by factor 2, so i switched to the PWM-Clock which is at factor 1.007 and now all five shutters work.

closed 2023-01-15 by NoI3ody

Nickduino, 2023-01-15:
Great to know!

Rodd8, 2023-01-29:
Hi,

I'm a beginner in the raspberry pi world, this project looks great but I'm stuck at this stage... I tried hayguen's clock test and it looks like I have a factor 2 in PCM too.
Everyone is talking about switching to PWM, but I have absolutely no idea how to do it, or what it means ^^
Can someone explain me how to switch from PCM to PWM?
Thanks

NoI3ody, 2023-01-29:
Hi Rodd8,

just try this code ;-) -t0 sets the PWM-Timer

`sudo systemctl stop shutters.service
sudo killall pigpiod
sudo pigpiod -t0
sudo systemctl start shutters.service`

Greetings

Moulders26, 2023-02-08:
> 
Thanks for this it was exactly what I needed to get mine working :) however it seemed to be needed after every reboot. I asked ChatGPT how to fix and it told me the below which helped me, just incase anyone else runs into this issue 

> To make the change stick after rebooting your system, you can add the command to your system startup scripts. Here's how:
> 1. Open the terminal and type sudo nano /etc/rc.local. This will open the rc.local file in the nano editor.
> 2. Add the following line before the exit 0 line: /usr/bin/pigpiod -t 0
> 3. Save and close the file using Ctrl + X, then Y, then Enter.
> 4. Make the file executable with the following command: sudo chmod +x /etc/rc.local
> Now, after every reboot, pigpiod will be automatically started with the -t 0 flag.

Nickduino, 2023-02-17:
That's a great use of ChatGPT