Nickduino/Pi-Somfy #153: Should that work in 433.92 MHz?

slashback100, 2023-03-23:
Hi,

I juste installed Somfy shutters. I see everywhere that the RF emitter should be adapted to work with a 433.42 MHz crystal, but my remotes are [AC123-01D](https://www.aokfrance.com/recepteur-de-telecommandes-7-1132_2-detail-2-tlcommandes-ac123.html) which works in 433.92 MHz. So my Somfy shutters are working in 433.92 MHz.
So I don't need to change the cristal.
Is the code supposed to work with this frequency? Or should the `sendCommand` function be adapted ?

Thanks for your help

Nickduino, 2023-03-23:
I have no idea...are you sure that remote is controlling RTS shutters from Somfy?

slashback100, 2023-03-23:
Well... it is supposed to... but indeed the more I dig, the more I have the feeling that it is not a Somfy... I'll check with the installer. 
I'll close this issue. Thanks

closed 2023-03-23 by slashback100