Nickduino/Pi-Somfy #154: TX pin is initialized high after startup

Scrath1, 2023-05-13:
Hello,
while viewing the signal of my transmitter using an SDR, I noticed that after startup of the raspberry pi, the TX pin of the transmitter is apparently initialized to high and a continuous signal is being sent.
[image]

The continuous transmission stops after the first command is manually sent. In my case I used a stop command.

Measuring the transmission pin with a multimeter confirms this. Until the first command is sent, the pin is at ~3.3V and afterwards drops to 0V.

I am using a Raspberry Pi 1 model B.

arthurdent2095, 2024-05-05:
I have always had the same problem and have a scheduled command at 2am to close a blind right after a reboot. 

But also, over time I am finding my transmitter boards stop working properly after about a year each.  It's as if they drift off frequency slightly, putting a finger on the board makes them work again, but they were fine from new. I think this behaviour with the GPIO does not help extend their life. 

If there is no fix yet, this time I am motivated to fix it in case it extends the life of the transmitter.

fbetil, 2026-09-28:
Hi,
I had the same problem with previous version. I fix the issue by adding a fake shutters (identifier 0x289621) and adding this line in operateShutters.py at line 521 (before starting webserver) : self.shutter.stop('0x289621').
I have just update to the latest version and I continue having the bug...

MichaelB2018, 2026-10-01:
Fixed in 2b7a190.

@arthurdent2095's analysis was correct: nothing drove the TX GPIO before the first command, so the pin stayed a floating input and the transmitter kept sending until `set_mode()` / `write()` finally ran inside the send path.

The fix is essentially @fbetil's workaround, minus the fake shutter:

- **`operateShutters.py` now drives the TX GPIO low right after the `Shutter` is constructed**, before the scheduler, web server and MQTT start. The transmitter therefore goes quiet at service start instead of at the first command. The RF backend already had a `set_idle_low()` for the CC1101 path; it is now also used at startup, and `CC1101Transmitter` forwards it to the underlying waveform transmitter.
- **On the lgpio path (Pi 5) the TX line is now claimed once and held.** Previously every transmission ended with `gpio_free()`, which hands the line back to the kernel as a floating input — so the startup fix alone would have been a no-op there. The claim is released on shutdown via `close()`.

On pigpio Pis the daemon retains the pin mode and level after the client disconnects, so the line stays driven low for the lifetime of the service.

A note on the hardware side: as @arthurdent2095 found, these cheap 433 MHz modules are genuinely unhappy when the data pin floats, and the decoupling capacitor mod on that electro-tech-online thread is still worth doing. This change at least stops Pi-Somfy from causing the condition.

Reopen if you still see a carrier after a reboot.

closed 2026-10-01 by MichaelB2018