Nickduino/Pi-Somfy #155: e07-m1101d compatibility

Sundancescore, 2023-05-20:
Hello and thanks for this great/huge project!

I have a e07-m1101d transmitter (Texas Instruments CC1101) that works in 433.42MHz but also on other frequencies. https://www.aliexpress.com/item/32816741903.html

Is this compatible with this project by default? Or do I need to send some extra commands to setup the oscillator to 433.42?

Scrath1, 2023-05-29:
I'm not the creator of this project and therefore not intimately familiar with it.
I don't believe this module is compatible with the current software.

Your transmitter module requires a connection over SPI, while the transmitter used here gets its transmit state directly from a GPIO pin. If you find a workaround for this, please let me know since I've been thinking about trying to replace my transmitter with a CC1101 as well. The transmitter used in this project has been very unreliable for me.

mattg2, 2023-06-02:
please have a look in the issue:
https://github.com/Nickduino/Pi-Somfy/issues/103

had it running several month, but it was kind of unreliable and I was not able to find the cause.
Switched then to https://github.com/rstrouse/ESPSomfy-RTS, which is ESP32 + CC1101 as simple as it could be.

Sundancescore, 2023-06-02:
> please have a look in the issue: #103
> 
> had it running several month, but it was kind of unreliable and I was not able to find the cause. Switched then to https://github.com/rstrouse/ESPSomfy-RTS, which is ESP32 + CC1101 as simple as it could be.

Wow, great! Thanks for the hints!

closed 2023-09-24 by Nickduino