Nickduino/Pi-Somfy #161: Test on RPI4

Zachary57-fr, 2023-09-23:
Hello,

Migrating from PI B+ to PI 4, commands are not working despite having no error on operateshutter log ( frame is correctly displayed on log but motor does not react ). Same transmitter works on PI B+

Since readme does not indicate test on Pi 4, this is not really an issue but can you please confirm if someone tried with PI 4 and got better results ?

Thank you

--------
Raspbian version : 

#uname -a 
Linux raspberrypi 6.1.54-v8+ #1682 SMP PREEMPT Wed Sep 20 15:19:57 BST 2023 aarch64 GNU/Linux

Pigpiod version : 
# pigpiod -v
79

GPIO port used :
27
--------

Zachary57-fr, 2023-09-24:
After some additionnal testing, I can see that some signal is send on GPIO port but malformed. 
I suspect that this come from pigpio.

Signal from a PI B+ : 
[image]

Signal from a PI4 ( same address and rolling code ) : 
[image]

Nickduino, 2023-09-24:
I would suggest to test your clock factor
https://github.com/Nickduino/Pi-Somfy/issues/136#issuecomment-1041907959

Zachary57-fr, 2023-09-25:
> I would suggest to test your clock factor [#136 (comment)](https://github.com/Nickduino/Pi-Somfy/issues/136#issuecomment-1041907959)

Hello,

No error from this test : 

`# ./test_delay
testing PWM clock
  delta = 50419, expected = 50000, ratio = 1.008380
  clock is OK
testing PCM clock
  delta = 50397, expected = 50000, ratio = 1.007940
  clock is OK`

Nickduino, 2023-09-25:
I don't have another idea at the moment

Maxvi33, 2023-10-05:
Hello @Zachary57-fr,
Did you find the solution ? 

bonne soirée,
Maxime

Zachary57-fr, 2023-10-07:
Hello,

I finally found the solution and I can confirm that it is working fine on PI4 8GB version 1.5
It seems that issue was due to conflict between pigpio ( needed for pisomfy ) and raspi-gpio ( installed by argonone case script ). If you are interested, you can find all details on this github issue : [https://github.com/joan2937/pigpio/issues/585](https://github.com/joan2937/pigpio/issues/585)

Have a nice week-end

closed 2023-10-07 by Zachary57-fr