Nickduino/Pi-Somfy #165: Not able to start PIGPIO (again, but on RPi5)

geohei, 2024-04-03:
I followed the instructions. However python-pigpio doesn't install correctly.
`geohei@rpi5:~/Pi-Somfy $ sudo apt-get install python-pigpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python-pigpio : Depends: python:any (< 2.8) but it is not installable
                 Depends: python:any (>= 2.7~) but it is not installable
E: Unable to correct problems, you have held broken packages.`
I belive (not sure ?!) that's the reason why I get:
`geohei@rpi5:~/Pi-Somfy $ sudo python3 ./operateShutters.py
Not able to start PIGPIO`
'geohei@rpi5:~/Pi-Somfy $ sudo pigpiod -l -m'
... shows nothing.
Any idea how to fix?

harebrainedkiwi, 2024-04-04:
The RPi5 hardware changed, where the GPIOs are now controlled through the RP1 southbridge, and as a result PIGPIO is not supported on the RPi5 - see the comment here from the PIGPIO developer stating that adding support "will be a huge task": https://github.com/joan2937/pigpio/issues/589#issuecomment-1798570009

I was able to modify the Pi-Somfy python code to instead use lgpio, which is compatible with the RPi5.  The generated gpio pulse timing is not as precise as using PIGPIO, but it seems to work ok with my Somfy shutter.  I have been testing it for the last few weeks, and I've only noticed one occasion where an automation didn't automatically close the shutter as expected - but I can't confirm if that was caused by lgpio timing or something else.

I guess the question for @Nickduino is whether this project is still under active development, and if it makes sense to add support for both PIGPIO and lgpio options (perhaps with an input flag to specify which to use)?  Or just fork the repo and replace PIGPIO with lgpio (which is essentially my current quick "hack" approach used to test it at the moment).

FYI, you'll also run into an issue with MQTT on RPi5 with the updated Paho library, but I see there is already mention of that here: https://github.com/Nickduino/Pi-Somfy/pull/164

geohei, 2024-04-06:
Thanks a lot for explaining.
Just to be sure ... I guess the RPi4 B is currently the latest model supported by the Pi-Somfy project, right?

Cbeir0, 2024-05-29:
Is it hard to modify pigpio to lgpio? I tried to do it aswell, but don't know where to start.

harebrainedkiwi, 2024-06-05:
Due to interest from others, I have created a fork with the basic changes required to replace pigpio with lgpio.  I do not currently have time to tidy up everything up, but the code is functional and working well on a Raspberry Pi 5 over the last couple of months.
https://github.com/harebrainedkiwi/Pi-Somfy