Nickduino/Pi-Somfy #167: Trying to program shutter - nothing happens..

arielgr, 2024-07-20:
Hi @Nickduino ,
First wanted to thank you for a great project !
I have an RPI v3, bought a transmitter board and changed the oscillator as needed..
After hooking up the hardware and installing the software, and running the web UI - when I finally get to the program stage, it doesn't seem to work.. e.g. when pressing the button on my existing remote the shutter jogs, but then after clicking program nothing happens.
I am not an electrician :) - but measured 5V on the oscillator pins (top/right) so seems like current is reaching it..
How would I go about understanding where this isn't working?

Many thanks for your help!
(attached a picture of the board)
[image]
[image]

bmoesdijk, 2024-08-15:
I experience the same issue. Any ideas?

kenji21, 2024-08-16:
Maybe : https://github.com/Nickduino/Pi-Somfy/issues/136

bmoesdijk, 2024-08-23:
Finally figured out what the problem was. Turns out Somfy uses 2 different radio interfaces one is called RTS and the other one IO. Although @Nickduino makes a very clear note to the RTS protocol I wasn't aware of the second one; Somfy IO. RTS uses 433.42MHz. IO uses a complete different frequency in the 868 MHz range. That would require a different transmitter and a different modulation. (e.g.: https://www.conrad.nl/nl/p/somfy-1822609-draadloze-module-2273656.html) So for everyone that wants to make use of this home automation project; Make sure your Somfy system makes use of RTS. You can find this out by looking at your transmitter. That should mention RTS or IO on the back. Another indication is the led (or leds) on your remote transmitter. When red, it uses RTS when green, it uses IO. I finally solved my automation issue on the pi by adding a relay module with 8 relays (https://www.kiwi-electronics.com/nl/8-kanaals-5v-relais-module-2983?search=KW-1614) and the above mentioned transmitter. With the 8 relay module I'm able to support up to 4 individual Somfy screens.