Nickduino/Pi-Somfy #168: 433,42 MHz Resonator not available for Europe?

s4thoste, 2024-12-26:
I am looking for a Oscillator with 433,42Mhz which is shipped to Germany/Europe but I cannot find a single one.
Does anyone know how to get it?

Nickduino, 2024-12-26:
I'm surprised you couldn't find one on eBay/AliExpress

Nickduino, 2024-12-27:
[image]

https://www.aliexpress.com/item/1005005625251557.html

s4thoste, 2024-12-27:
Unfortunately, this one can also not shipped to Germany

Nickduino, 2024-12-27:
Try contacting the store?

[image]
https://tpake.aliexpress.com/store/912479009?spm=a2g0o.detail.0.0.60afPudCPudCb4

tfcarlin, 2024-12-27:
I have a bunch of components from my exploration of Somfy radios. I'm happy to ship these all to you if interested.
[image]

s4thoste, 2024-12-27:
This would be fantastic for me! 
Just Tell what I have to do for it

tfcarlin, 2024-12-27:
Let me know how to get in touch with you and we'll coordinate there.

s4thoste, 2024-12-28:
> Let me know how to get in touch with you and we'll coordinate there.

Perfect, you can send me an E-Mail and I'll provide you with my details :-)

s4thoste@gmx-topmail.de

Nickduino, 2024-12-28:
> I'm happy to ship these all to you if interested.

What a nice gesture!!!

Do you have 433.42 MHz crystals, though?

tfcarlin, 2024-12-28:
The only crystals are the ones soldered to the boards.

Nickduino, 2024-12-29:
But the original ones at 433.92, right?
So that won't solve @s4thoste's issue.

But I see you have plenty of CC1101, which might work with Pi-Somfy (not sure, you'd have to learn how to work with these) and will absolutely work with an ESP and https://github.com/rstrouse/ESPSomfy-RTS

tfcarlin, 2024-12-29:
If you zoom in you'll see they are 433.4. I replaced the originals...sort of foggy on the details so many years later. I then switched to CC1101 solutions which worked well on a RaspPi as well. I think there will be something in the batch that should work...and not a bad deal for the price of shipping.

Nickduino, 2024-12-29:
> If you zoom in you'll see they are 433.4.

All the better.

> not a bad deal for the price of shipping.

No, it's very generous of you

You should check out ESPSomfy-RTS